TheThingsIndustries/lorawan-stack-docs #1142: Some readers think that certain pages have little or no content

pradeeka7, 2023-07-17:
#### Summary

Some readers think that pages similar to this one (https://www.thethingsindustries.com/docs/gateways/models/) have a lack of content or no content. However, the content is actually there organized as sub-sections and accessible from the left menu.

[image]

#### Why do we need this ?

...

#### What is already there? What do you see now?

To easily find the content of the page.

#### What is missing? What do you want to see?

Add a card for each sub-section like this. 

[image]

#### How do you propose to document this?

However, I understand that if a page has a large number of sub-sections, this may not be possible. Therefore, we have to think about an alternative solution (e.g. for the gateway models page, we can add 'This page contains guides with instructions for connecting popular LoRaWAN® gateways. Simply select the gateway that you want to learn about from the left menu).

#### Can you do this yourself and submit a Pull Request?

Not yet, let's finalize a solution @nejraselimovic @benolayinka

nejraselimovic, 2023-07-23:
We already had cards there and removed it because it was too crowded. I think the easiest and best solution is to just reword this sentence and point people to look at the left menu.

pradeeka7, 2023-07-23:
> I think the easiest and best solution is to just reword this sentence and point people to look at the left menu.

let's do it.

KrishnaIyer, 2023-08-01:
Great @pradeeka7 can you come up with a pull request?

pradeeka7, 2023-08-05:
@KrishnaIyer Yes, I am working on it.

KrishnaIyer, 2023-08-10:
closed by #1151

closed 2023-08-10 by KrishnaIyer