TheThingsIndustries/lorawan-stack-docs #1144: ConnectedTimeBucket API response missing from docs (and probably others too)

benolayinka, 2023-07-18:
#### Summary

While making a video about the NOC API I realized this link is broken and the response is not documented

[image]

@johanstokking sent the description, is it missing from the api files or is there something that's not building?

`
// Number of entities connected in a time bucket.
message ConnectedTimeBucket {
  google.protobuf.Timestamp time = 1 [(validate.rules).timestamp.required = true];
  uint32 count = 2;
}

// Number of entities connected.
message ConnectedTimeBuckets {
  repeated ConnectedTimeBucket hourly_buckets = 1;
  repeated ConnectedTimeBucket daily_buckets = 2;
  repeated ConnectedTimeBucket weekly_buckets = 3;
}
`
In general our API docs are pretty clunky and we should probably consider refactoring how we build/present them.

johanstokking, 2023-07-18:
@benolayinka you can insert the missing message(s) on the page, see https://github.com/TheThingsIndustries/lorawan-stack-docs/blob/master/doc/content/reference/api/network_operations_center.md. Please check any other missing messages if any, other than these two.

nejraselimovic, 2023-11-20:
I'll close this issue since missing messages have been added in https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/1185 and https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/1201. If we see something missing we can open another issue, and for refactoring as well.

closed 2023-11-20 by nejraselimovic