TheThingsIndustries/lorawan-stack-docs #1145: Replace header and footer with ttui components

mjamescompton, 2023-07-19:
#### Summary

Replace header and footer with ttui components

#### Why do we need this?

Style improvements

#### What is already there? What do you see now?

Fully functioning header and footer

#### What is missing? What do you want to see?

Include ttui components after our internal discussions and feature request to include search in the ttui header component 

#### Can you do this yourself and submit a Pull Request?

yes, but @pierrephz can handle it

pierrephz, 2024-01-02:
@KrishnaIyer do you have thoughts here? if you put yourself in the user shoes?

KrishnaIyer, 2024-01-03:
> The Things Stack
Device
Gateways
Intergrations

I'm not sure this would work. Devices/Gateways (maybe integrations) are not logical sub-sections of The Things Stack. Those sections include generic content about these which are not really part of TTS.

Taking a few steps back, do we need to make the docs header the same as tti.com?

My experience is that they can be different since they are meant for different purposes.
Take Stripe for example https://stripe.com/en-gb-nl and https://stripe.com/docs.