TheThingsIndustries/lorawan-stack-docs #1147: Improve the page 'Server Addresses' to easily find the server addresses

pradeeka7, 2023-07-23:
#### Summary

Improve the page '[Server Addresses](https://www.thethingsindustries.com/docs/the-things-stack/concepts/server-addresses/)' to make it easier to find server addresses, as it is currently a bit difficult to find server addresses for some clusters, man1 cluster, for example.

#### Why do we need this ?

A reader's feedback mentions that he/she can't find the server addresses for North America.

#### What is already there? What do you see now?

Server addresses with only randomly selected clusters as examples.

#### What is missing? What do you want to see?

[image]

#### How do you propose to document this?

At the top of the page, add a drop-down list to select the cluster first, and based on the selection, the server addresses will dynamically change.

#### Can you do this yourself and submit a Pull Request?

Yes, but I am waiting for Nejra's ( @nejraselimovic ) idea.

nejraselimovic, 2023-07-25:
We already have this implemented, for example here https://www.thethingsindustries.com/docs/the-things-stack/cloud/addresses/, we just need to add it on these pages. Do you want to do it or should I?

pradeeka7, 2023-07-27:
@nejraselimovic  - assign this to me.

pradeeka7, 2023-08-31:
Closed because the related PR was merged.

closed 2023-08-31 by pradeeka7