TheThingsIndustries/lorawan-stack-docs #1148: Large volume of text in Regional Parameters page

pradeeka7, 2023-07-27:
#### Summary

The Regional Parameters page has a large volume of text and needs to be reduced.
https://www.thethingsnetwork.org/docs/lorawan/regional-parameters/

#### Why do we need this ?

To find key parameters for each region easily.

#### What is already there? What do you see now?

The regional parameters that are required for The Things Certification.

#### What is missing? What do you want to see?

...

#### How do you propose to document this?

Remove the text that is not required for the exam.

#### Can you do this yourself and submit a Pull Request?

Yes, @nejraselimovic please assign this to me.

KrishnaIyer, 2023-08-10:
This is the wrong repository I think. The page you are referring to in on the TTN docs.

pradeeka7, 2023-08-10:
@KrishnaIyer Oops! Thanks for pointing that out. I will close this issue now.

closed 2023-08-10 by pradeeka7