TheThingsIndustries/lorawan-stack-docs #1166: Document the JoinEUI/AppEUI selection insights for the devices when using The Things Join Server

ymgupta, 2023-08-26:
#### Summary

The Things Join Server is a stand-alone LoRaWAN® Join Server. The Things Join Server can be used by device makers to provision end devices to be used by all distributions of The Things Stack as well as any other standards compliant LoRaWAN Network Servers. The registration of end devices frequently leads to user confusion regarding the proper utilization of JoinEUI with The Things Join Server.

#### Why do we need this ?

To enhance users' understanding of JoinEUI usage while using The Things Join Server. 

#### What is already there? What do you see now?

[The Things Join Server
](https://www.thethingsindustries.com/docs/the-things-stack/host/join-server/)
#### What is missing? What do you want to see?

Add the information about the JoinEUI range when using The Things Join Server in The Things Join Server documentation. 

#### How do you propose to document this?

**Insights:** 
As per our earlier context, when using The Things Join Server, the AppEUI/JoinEUI should be taken from the `70B3D57ED0000000/40` block, i.e., in the range from `70B3D57ED0000000` to `70B3D57ED0FFFFFF`. 

#### Can you do this yourself and submit a Pull Request?

No, @nejraselimovic

KrishnaIyer, 2023-10-03:
> As per our earlier context, when using The Things Join Server, the AppEUI/JoinEUI should be taken from the 70B3D57ED0000000/40 block, i.e., in the range from 70B3D57ED0000000 to 70B3D57ED0FFFFFF.

Where did this requirement come from? Afaik, customers cannot arbitrarily choose an EUI from this range. This range is owned by TTI and we are responsible for allocating EUIs from it for various purposes.

In general, there are two cases
1. **TTI hosted The Things Join Server:** Provisioners may choose `70B3D57ED0000000` if they are provisioning their devices on TTI hosted TTJS and they don't have a EUI themselves. If they want other Join EUI(s) to be configured on TTJS, they must discuss with TTI since we have to check if we can use that EUI and/or update some settings.
2. **Externally hosted The Things Join Server:** Provisioners must choose a JoinEUI that is owned by them, acquired via IEEE.

cc: @johanstokking

johanstokking, 2023-10-04:
@KrishnaIyer yes exactly right.

@ymgupta TTJS is currently only `70B3D57ED0000000/64`, not a range.

PR #1186 (2023-10-13): doc: Note about JoinEUI range
#### Summary
Closes #1166 

#### Notes for Reviewers
I'm not sure if I understood this correctly so I'm also not sure if this PR covers the issue.. So my question - when a user uses TTJS, there is a pool of JoinEUIs (mentioned in PR), which are assigned to devices upon device provisioning and need to be used when registering those devices on TTS? And users are confused because this range isn't clearly stated in our docs?

#### Checklist
- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build u [... cut]
closed 2023-10-13 by nejraselimovic