TheThingsIndustries/lorawan-stack-docs #1167: Broken URLs on the 'Tektelic Kona Micro IoT LoRaWAN Gateway' page

pradeeka7, 2023-08-28:
#### Summary

There are some broken URLs on the Tektelic Kona Micro IoT LoRaWAN Gateway page.
https://www.thethingsindustries.com/docs/gateways/models/tektelickonamicro/

#### Why do we need this ?

The linked pages do not exist on the Tektelic side.

#### What is already there? What do you see now?

The following links are broken.
https://tektelic.com/catalog/kona-micro-iot-gateway
https://support.tektelic.com/portal/en/kb/articles/konaft-instructional-videos

#### What is missing? What do you want to see?

...

#### How do you propose to document this?

...

#### Can you do this yourself and submit a Pull Request?

Yes, but I want to contact someone at Tektelic to find the correct resources.
@nejraselimovic please assign this to me.

nejraselimovic, 2023-08-29:
Could you also take this one https://github.com/TheThingsIndustries/lorawan-stack-docs/issues/856 as it is related to the same gateway?

pradeeka7, 2023-08-30:
> Could you also take this one #856 as it is related to the same gateway?

Yes

KrishnaIyer, 2023-10-01:
@pradeeka7: Please pick this and https://github.com/TheThingsIndustries/lorawan-stack-docs/issues/856 up  and try to complete them in the `Oct 2023` milestone.

pradeeka7, 2023-10-12:
> Please pick this and #856 up and try to complete them in the `Oct 2023` milestone.

I am afraid I won't be able to take over this issue because I don't have a Tektelic Kona Micro gateway for testing. However, it seems there's no need to apply any fixes to the existing documentation.

closed 2023-10-13 by nejraselimovic