TheThingsIndustries/lorawan-stack-docs #1188: Minor additions to the Subscription and Billing Management documentation

ymgupta, 2023-10-05:
#### Summary

Add the below queries of The Things Stack Cloud users to `The Things Stack Subscription and Billing Management portal`:

- How can I downgrade my support tier subscription?
- What are the steps to delete the Subscription and Billing Management Account?

#### Why do we need this?

To assist The Things Stack Cloud users with the common queries on the Subscription and Billing portal. 

#### What is already there? What do you see now?

[Subscription and Billing Management
](https://www.thethingsindustries.com/docs/the-things-stack/cloud/billing-and-subscription-management/)

#### What is missing? What do you want to see?

Add the below queries to the FAQs of Subscription and Billing Management documentation. 

  - How can I downgrade my support tier subscription?
  - What are the steps to delete the Subscription and Billing Management Account?

#### How do you propose to document this?

1. How can I downgrade my Support subscription?

Please follow the process mentioned in the guide [managing your subscriptions, support, and billing plans](https://www.thethingsindustries.com/docs/the-things-stack/cloud/billing-and-subscription-management/#manage-your-subscriptions-support-and-billing-plans) for unsubscribing the support plan.

2. What are the steps to delete the Subscription and Billing Management Account?

Currently, we don't have this option in the Subscription and Billing Management account.

@rish-c, could you please your insights here?

#### Can you do this yourself and submit a Pull Request?

No, @nejraselimovic

rish-c, 2023-10-09:
Some topics that we need documentation for - 
* Adding or editing a payment method - update your existing payment method
* Downgrade (Cloud tiers, Support)
* Setting your billing email

Also please attach the latest screenshots from subscription management. 

cc @mjamescompton

mjamescompton, 2023-10-10:
> Setting your billing email

There is a discussion here, once this is done I should get into the routine of updating this doc as part of the task 
https://github.com/TheThingsIndustries/subscription-management/issues/180

nejraselimovic, 2023-10-13:
Hey guys, I've been working on this and tried to "buy" discovery subscription, but when I enter my personal and billing info, I get this. Do you know what could be causing it?
[image]

nejraselimovic, 2023-10-13:
Hey guys, I've been working on this, so I tried to "buy" discovery subscription but when I enter my personal and billing info I get this error. Do you know what might be causing it?
[image]

nejraselimovic, 2023-10-13:
Also, "Install command line interface" link on this step is broken 
[image]

nejraselimovic, 2023-10-13:
CLI-related link in the email I got is broken as well

[image]

rish-c, 2023-10-13:
Thanks @nejraselimovic. I see this as well from people trying to sign up, it gives an error and then asks people to reach out to support. It would be good to give more context on the process.

nejraselimovic, 2023-10-13:
I also have a question about admin password -> when creating an admin user, there is no step to configure a password. When the Cloud tenant is finally launched, it redirects you to log in the Console. Upon login, it asks for username and password. Username is the admin name you set before, but which password do you use? I got some temporary tokens in my email when launching the tenant, but couldn't log in with any of those. Is it intended to have to change password when first time logging in or what? I think
I missed something here

mjamescompton, 2023-10-16:
@nejraselimovic 

> Is it intended to have to change password when first time logging in or what?

yes, there should be an email asking you to setup you admin account on the stack. We are aware there are a lot of emails when you setup and unless you read every word it not that straight forward. As the tenant is created by sending metadata to stripe we can not safely pass the password to the stack. We have spoken about adding this feature. This process will be changed soon to be created via the Network registry which may give us the ability to create password in this form. @johanstokking did you make this possible via NR? Are there still issues around doing this? 

> I got some temporary tokens

This is used to give you access to the tenant setup form which you can access directly in the billing flow. The email is there to give you access to the form if you don't know what the tenant id etc should be, or if you want to pass that responsibility to someone else. Its also good to know, because of this access token you don't need to logged into the subscription management dashboard to configure the tenant, if you have the url you can share it.

> I missed something here.. I managed to log in only when I followed steps for changing password. Note that I used my work email which was already connected to a username 'snejra' that I generally use on community network, and now I created 'nejraselimovic' with the same email addr for testing purposes. Is it then the same password as for 'snejra'?

No, cloud hosted and the community network are not connected.

> Is there any way to make this error more straightforward, so that user knows what the issue is? I kept wondering if its diacritic characters in my last name and address, + in my phone number, etc. When I unblocked my card for internet payments it worked, but I think it could be more suggestive.

We have a collection of error we display, we can add more. In you case if you can use the inspector and check the Network tab you should be able to see what the exact error from the server is. If you create an issue and include what you think the error message should be I can include it.

> Also, do you have any way for me to "test" this whole process, like purchasing deployment and support options other than Discovery and Free, upgrading/downgrading and other stuff, without actually involving my personal credit card? Is there even need to do this? I'm not sure since I never went through that part of the platform :)

You can use the preview servers.

The whole billing flow is connected here all the way to `staging1.cloud.thethings.industries`
Start here: https://preview.thethingsindustries.com/stack/plans/ and it should redirect you to 
https://preview.accounts.thethingsindustries.com/

Use the stripe testing card in the checkout https://stripe.com/docs/testing. In most cases just use the card number `4242 4242 4242 4242` or if you want to test other scenarios there are cards listed  for testing 3d secure, failed payments etc

> CLI-related link in the email I got is broken as well

Can you make a stack issue for this.

johanstokking, 2023-10-17:
> This process will be changed soon to be created via the Network registry which may give us the ability to create password in this form. @johanstokking did you make this possible via NR? Are there still issues around doing this?

TTNR leaves the password generation up to TTSC, hence it will be a magic link sent via email, like now.

The reason we can't prompt the password in account setup, is that password managers will not save it under `TENANT.eu1.cloud.thethings.industries`, where it should be, for login later. There's no universal form hinting we can use so that the plethora of password managers (third party and browser/OS native) will correctly store custom acceptable domains. So, we have to ask the user to enter the password while visiting `TENANT.eu1.cloud.thethings.industries`.

nejraselimovic, 2023-10-24:
> We have a collection of error we display, we can add more. In you case if you can use the inspector and check the Network tab you should be able to see what the exact error from the server is. If you create an issue and include what you think the error message should be I can include it.

I just reproduced this and I got [error 402](https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/402) in the Network tab. I guess it's not specific enough to display any other error on our side

PR #1190 (2023-10-25): doc: Update TTS Plans info
#### Summary
Closes #1116
Closes #1188

#### Checklist
- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is marked using the `new-in-version` shortcode, according to the guidelines in [CONTRIBUTING](CONTRIBUTING.md).
- [x] Style Guidelines: Documentation obeys style guidelines in [ [... cut]
closed 2023-10-25 by nejraselimovic