TheThingsIndustries/lorawan-stack-docs #1192: Add the scenario `unable to access the end device` due to incorrect server addresses

ymgupta, 2023-10-06:
#### Summary

Sometimes, The Things Stack users will mistakenly configure the end device server addresses incorrectly, leading to access issues with the end device in The Things Stack Console and showing as `Other cluster`. Below is the example scenario with the incorrect server address.
[image]

To reaccess the end device, it should be configured with the correct server addresses. Please add this scenario and resolution steps to the [device troubleshooting guide](https://www.thethingsindustries.com/docs/devices/troubleshooting/).  

#### Why do we need this ?

To assist users in identifying and resolving common errors with device access in The Things Stack.

#### What is already there? What do you see now?

[Troubleshooting Devices](https://www.thethingsindustries.com/docs/devices/troubleshooting/)

#### What is missing? What do you want to see?

Add the scenario `unable to access the end device in the console` to the device troubleshooting guide. 

#### How do you propose to document this?

**Insights:** 

- Users can either update the end device's server address by accessing the device's general settings or delete the end device and then re-create it in The Things Stack.
- To access the end device's general settings for updating the server addresses or deleting the device from The Things Stack Console, users can use the below similar URL and access it into their web browser where they have logged into The Things Stack: 

**Example URL to access the device's general settings page:**

```
https://<thethings.example.com>/console/applications/<application-id>/devices/<device-id>/general-settings
```

**Note:** Make sure to replace `<thethings.example.com>, <application-id>, and <device-id>` accordingly. 

#### Can you do this yourself and submit a Pull Request?

No, @nejraselimovic

nejraselimovic, 2023-11-19:
@ymgupta I see you put links to some images, but I cannot access them. Can you pls put images directly in the issue template next time

nejraselimovic, 2023-11-19:
Btw which server address are we talking about, NS, AS, JS, or any one of those can create this error?

ymgupta, 2023-11-21:
@nejraselimovic , The reported behavior could occur if all three server addresses (AS/NS/JS) are configured with incorrect server addresses. If any one or two server addresses (AS/NS/JS) are configured incorrectly, it won't cause the reported behavior.

closed 2023-11-26 by nejraselimovic