TheThingsIndustries/lorawan-stack-docs #1200: (Very) minor improvements to CONTRIBUTING.md

Kostas-F, 2023-10-19:
### Summary

Some minor problems I spotted with the md file that would be quick to fix. All trivial, but sorted in my perceived order of "importance":

1. Line 5: The link to the forum category is dead. I believe the correct url now is `https://www.thethingsnetwork.org/forum/c/ttn-network/v3/90`

2. Line 9: The site referenced as the location of user documentation is `thethingstack.io`. It properly redirects to the thethingindustries docs location so no problems there. However correcting the user facing link would help with dissambiguiation. 

3. Also Line 9: Missing parenthesis in the docs repository link.

Kostas-F, 2023-10-20:
This is in reference to the CONTRIBUTING.md in The Things Network repository, I just realized the issue opened in this repository when I clicked on the documentation request. I hope it is not in the wrong place

nejraselimovic, 2023-11-12:
Hey @Kostas-F, thanks for reporting, this should've been submitted to TTN and TTI `lorawan-stack` repos indeed, but never mind, I've submitted PRs to fix this on both places:

https://github.com/TheThingsIndustries/lorawan-stack/pull/3951
https://github.com/TheThingsNetwork/lorawan-stack/pull/6703

Kostas-F, 2023-11-13:
Hi @nejraselimovic I apologize for the confusion, thank you for taking the time.

closed 2023-11-19 by nejraselimovic