TheThingsIndustries/lorawan-stack-docs #1229: Minor improvements in the hosting TTS on AWS ECS documentation

Siva-7482, 2023-12-14:
#### Summary

We have identified minor improvements in a few sections in hosting the TTS on AWS ECS documentation.

#### Why do we need this ?

To assist users in hosting The Things Stack on AWS ECS with ease.

#### What is already there? What do you see now?

Hosting TTS on [AWS ECS](https://www.thethingsindustries.com/docs/getting-started/aws/ecs/) documentation.

#### What is missing? What do you want to see?

### Documentation improvements specific to each section:
  - **Who is it for?**
    - Update with more information on the minimum number of devices, gateways, and deployment size of the users in numbers to consider for hosting the TTS on AWS ECS.
    - Add optional concepts related to Prometheus, Grafana, and OpsGenie as nice to have context.
    Ref: [Who is it for?](https://www.thethingsindustries.com/docs/the-things-stack/host/aws/ecs/#who-is-it-for)
  - **Containers:**
    - Update about the UDP container requirements in the deployment according to the latest changes. In [TTS v3.26.0](https://www.thethingsindustries.com/docs/the-things-stack/host/aws/ecs/changelog/#ecs-templates-1) of The Things Stack Cloud, the Gateway Server with a Network Load Balancer is using for the UDP gateways by removing the UDP gateway server service.
    Ref: [Containers](https://www.thethingsindustries.com/docs/the-things-stack/host/aws/ecs/architecture/#containers)
  - **Prerequisites:**
    - Update the NOC OAuth Client ID and Secret requirements similar to the Console and Device Claiming Server in the prerequisites section.
    - Context on the certificate requirement of the below two points.
        - Configuration and Certificates for Backend Interfaces interop (optional).
        - Configuration and Certificates for Packet Broker peering (optional).
    - Add information on the alerting and monitoring requirements.
Ref: [Prerequisites](https://www.thethingsindustries.com/docs/the-things-stack/host/aws/ecs/prerequisites/)
   
  - **Deployment:**
     - Update the below information in the [ECS cluster](https://www.thethingsindustries.com/docs/the-things-stack/host/aws/ecs/deployment/#ecs-cluster) section with the latest UI changes in the AWS ECS console.
        ```
        Under Amazon ECS ARN and resource ID settings the new ARN and resource ID format needs to be enabled for container instances, services and tasks. If you plan to deploy The Things Stack on container instances, AWSVPC Trunking needs to be enabled.
        ```
   - **Database Operations:**
      - Add Identity Server initialization section with the command `tti-lw-stack, is-db, migrate` 
      - Add DCS OAuth client creation command.
      Ref: [Database Operations](https://www.thethingsindustries.com/docs/the-things-stack/host/aws/ecs/database-operations/)
 
  - **Monitoring:**  
      - Add example instructions for configuring the grafana dashboards with the Prometheus deployed by The Things Stack.
      Ref:  [Monitoring](https://www.thethingsindustries.com/docs/the-things-stack/host/aws/ecs/monitoring/)

- **Updating:**
     - Add NOC database migration command in the [migrating](https://www.thethingsindustries.com/docs/the-things-stack/host/aws/ecs/updating/#migrations) section.

- **Best Practices:**
     - Add insights on the infrastructure optimization for the user deployments based on their deployment size, i.e., number of devices and gateways.
     Ref: [Best Practices](https://www.thethingsindustries.com/docs/the-things-stack/host/aws/ecs/best-practices/)

#### How do you propose to document this?

...

#### Can you do this yourself and submit a Pull Request?

No, @KrishnaIyer

KrishnaIyer, 2024-04-04:
@Siva-7482: As discussed earlier, if you have all the required information, can you please open a pull request yourself? cc: @ymgupta

Siva-7482, 2024-04-10:
@KrishnaIyer,

We do not have the required information to file the pull request here. And it requires insights from your end. Could you please provide us with your insights for the topics mentioned above? Accordingly, we will open a PR with the changes.

KrishnaIyer, 2024-04-10:
Ah I see. Ok

KrishnaIyer, 2024-07-01:
Let's pick this up during the conference month.

KrishnaIyer, 2025-01-29:
This item is slightly outdated. I'll come up with a replacement.

closed 2025-01-29 by KrishnaIyer