TheThingsIndustries/lorawan-stack-docs #1241: Add ELSYS ERS series.

LDannijs, 2024-01-09:
#### Summary

Add documentation for onboarding the ELSYS ERS series to The Things Stack

#### Why do we need this ?

This device is very popular, so adding instructions to the documentation will help users.

#### What is already there? What do you see now?

...

#### What is missing? What do you want to see?

...

#### How do you propose to document this?

I have personally tested the device and know how to word it.

#### Can you do this yourself and submit a Pull Request?

Yes

LDannijs, 2024-01-09:
Should have discussed first but @pradeeka7 will do this one.

nejraselimovic, 2024-01-14:
@pradeeka7 @LDannijs just a note regarding issues and PRs -> when you open an issue, it automatically gets labeled with "needs/triage" label which means team should breakdown the issue and assign someone to pick it up. Since you guys already know that that you will be working on this and know how to do it, that label should be removed upon issue creation, and replaced with "documentation" label. You should also add 3 things - yourself as an assignee, someone for a review, and a milestone. Then, when you create a PR for that issue, in PR description under Summary put "Closes #issue_number", so that when you merge that PR, Github automatically recognizes and closes corresponding issue. It helps the team to keep the repo clean

PR #1245 (2024-01-15): doc: Add Elsys ERS Lite

#### Summary

This document outlines onboarding ELSYS ERS Lite onto The Things Stack.
Closes #1241 

#### Screenshots

[image]

#### Changes

- ...
- ...

#### Notes for Reviewers

@nejraselimovic / @KrishnaIyer

#### Checklist

- [X] Scope: The referenced issue is addressed, there are no unrelated changes.
- [X] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is ma [... cut]
closed 2024-01-15 by nejraselimovic