TheThingsIndustries/lorawan-stack-docs #1256: JSON device reference does not mention session.keys.session_key_id

dajtxx, 2024-01-29:
I have been moving devices between applications by exporting the device JSON via the CLI and then feeding that JSON back to the CLI dev create command. I am trying to keep the device session alive.

I removed all the fields not mentioned on the [JSON file reference](https://www.thethingsindustries.com/docs/devices/adding-devices/adding-devices-in-bulk/device-json/) page and received an error mentioning the `session.keys.session_key_id` field. When I put that field back the device creation worked.

I think this field needs to be added to the JSON reference table.

KrishnaIyer, 2024-01-30:
I have a PR coming that documents this. Will get back here when that's merged.

dajtxx, 2024-01-31:
More experimentation shows some fields not mentioned on the reference page are ignored so can be left in the JSON, and others are accepted.

For example, I left in things like the servers and created time at the top level and they were ignored which is fine.

I also left the `formatters` object in and the new device had those formatters carried over despite the JSON ref not mentioning this field/object. Using the whole formatters object during `dev create` a lot easier than using the 4 different flags to the `dev set` command so if this is intended behaviour it is worth mentioning.

dajtxx, 2024-02-01:
I think the page should also mention that `locations` and `formatters` are accepted. It might even be better to say what *should not* be included in the JSON.

dajtxx, 2024-02-05:
Further to my comment above, I have found that exporting with `--all` and then removing the application id, dev_addr (from the ids object) - both neccessary, and I also removed recent uplinks and downlinks although I don't know if this is required, worked much better.

I moved 2 devices between apps following the JSON reference and the devices do uplink, but very sporadically. I moved another 2 devices (all the same type) using the method described in this comment and these two devices missed 1 uplink but then carried on with regular uplinks.

So following the JSON reference to the letter does not seem to work for all devices. It also means you lose device-specific formatters.