TheThingsIndustries/lorawan-stack-docs #1257: Document how to use the CLI to move a device between apps while keeping the session alive

dajtxx, 2024-01-29:
#### Summary

It is not obvious how to move a device from one app to another, nor is it obvious how to keep the session alive. I pieced together the process with experimentation and use of the JSON file reference page.

It would be useful to have a page with example commands on how this can be done.
...

#### Why do we need this ?

Wanting to move devices between applications is sometimes desirable and having to reboot a device is not convenient if it is not necessary.

#### What is already there? What do you see now?

The JSON file reference is good, but I think it is missing a field as reported in another issue.

The CLI documentation is mostly just a web rendering of the in-built help but does not explain how to use it.
...

#### What is missing? What do you want to see?

The process I have come to, and which I don't know to be correct is:

Export a device.

```
ttn-lw-cli dev get <app-id> <dev-id> --name --description --lorawan-version --lorawan-phy-version --frequency-plan-id --supports-join --root-keys --session >dev.json
```

Delete the device from the original app.

Delete all the fields from the JSON file not mentioned in the JSON file reference page.

Create the device in the new app.

```
ttn-lw-cli dev create --application-id <new-app-id> <dev.json
```

#### How do you propose to document this?

Either:

* Create a new page describing how to do this process.
* Add a `dev move` command to the CLI and appropriate end point to the API to perform all the necessary steps in one go.

#### Can you do this yourself and submit a Pull Request?

No.

KrishnaIyer, 2024-01-30:
Thanks. We'll look into this request.

dajtxx, 2024-02-05:
See my comments on [this issue](https://github.com/TheThingsIndustries/lorawan-stack-docs/issues/1256) for a more reliable way of moving devices between applications.

KrishnaIyer, 2024-02-29:
So moving devices between applications is currently via the migration tool https://github.com/TheThingsNetwork/lorawan-stack-migrate.

If you run into issues, please create an issue there.

closed 2024-02-29 by KrishnaIyer