TheThingsIndustries/lorawan-stack-docs #1270: Missing tereform script on document

pospospos2007, 2024-02-21:
Hi, can you please provide the tereform script on https://www.thethingsindustries.com/docs/the-things-stack/host/kubernetes/azure/deployment/
which is missing for a long time. I'm currentlly usiung open source verison the thingstack and deploy it to AKS.

Thanks,
Albert

KrishnaIyer, 2024-02-22:
Hey thanks for reporting this. We can find a way for you to access the scripts. 

However, it's important to note that the azure k8s deployment only works with The Things Stack Enterprise versions and not with the open source distribution. This is since the deployment requires replication and that's not supported in the open source model.

pospospos2007, 2024-02-22:
No problem, can you share the script ? I will apply for  Enterprise versions.

pospospos2007, 2024-02-25:
Kindle reminder, is there any upate on this? I'm still waiting for the script. Let me know if need anything. Happy weekend!

KrishnaIyer, 2024-03-06:
Here is the example infrastructure.

[lw-stack-k8s-infra-azure.zip](https://github.com/TheThingsIndustries/lorawan-stack-docs/files/14510118/lw-stack-k8s-infra-azure.zip)

Please note that you will need an enterprise license for this to work.