TheThingsIndustries/lorawan-stack-docs #1302: Add error `server_address_mistmatch` with The Things Stack CLI to Troubleshooting CLI documentation

Siva-7482, 2024-04-06:
#### Summary

The Things Stack CLI can be used for advanced actions. Accessing The Things Stack CLI requires the correct CLI configuration with the server addresses. After that users can perform actions on the entities that are registered on the same server address. If the users try to perform actions on the other server address, then the error `error:cmd/ttn-lw-cli/commands:end_device_server_address_mismatch ` will be observed. Please document the error and resolution in [The Things Stack CLI troubleshooting](https://www.thethingsindustries.com/docs/the-things-stack/interact/cli/troubleshooting/)guide.

#### Why do we need this ?

To assists users in resolving the issues on their own while using The Things Stack CLI. 

#### What is already there? What do you see now?

[Troubleshooting CLI
](https://www.thethingsindustries.com/docs/the-things-stack/interact/cli/troubleshooting/)

#### What is missing? What do you want to see?

Add the error `error:cmd/ttn-lw-cli/commands:end_device_server_address_mismatch (Network/Application/Join Server address mismatch)` to the Troubleshooting CLI documentation. 

#### How do you propose to document this?

**Insights:** 

- Generally, the error `error:cmd/ttn-lw-cli/commands:end_device_server_address_mismatch (Network/Application/Join Server address mismatch)` occurs when the user tries to perform the actions on entities (i.e., devices, gateways, users, etc) using the CLI with the incorrect server address configured in the CLI configuration file. For example, if the user configured the CLI with the `eu1` cluster server address and if they try to access the entities on the `au1` cluster, then the error will be observed due to the mismatch of server addresses in the configuration file.
- While configuring the server address in the CLI configuration, users need to ensure the correct cluster address to perform any operations on the entities registered in that specific cluster. So if the users want to perform operations on the entities registered in the `eu1` cluster then they need to ensure the CLI configuration with the `eu1` cluster.
- Also, it is important to note that the Identity Server address in the CLI configuration should always be the `eu1` cluster irrespective of the cluster that the user trying to access the entities registered on.

#### Can you do this yourself and submit a Pull Request?

No, @nejraselimovic

KrishnaIyer, 2024-04-08:
@Siva-7482: Why don't you just create a PR for this? All the required information is there in the issue description. cc: @ymgupta

Siva-7482, 2024-04-10:
Sure, @KrishnaIyer. Please review the below PR.

Ref: https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/1309

closed 2024-04-21 by nejraselimovic