TheThingsIndustries/lorawan-stack-docs #1332: Document connecting the Dragino LPS8 gateway in LoRa Basics Station mode to The Things Stack

peetharavi, 2024-04-20:
#### Summary

We can connect Dragino LPS8 Gateway to The Things Stack using UDP and LoRa Basics Station modes. However, we don't have a documentation guide for connecting the Dragino LPS8 gateway in LoRa Basics Station mode to TTS.

#### Why do we need this?

To assist users in connecting the Dragino LPS8 Gateway to The Things Stack using LoRa Basics Station. 

#### What is already there? What do you see now?

[Connecting Dragino LPS8 to TTS in UDP mode](https://www.thethingsindustries.com/docs/gateways/models/dragino-lps8/)

#### What is missing? What do you want to see?

Document how to connect the Dragino LPS8 Gateway to The Things Stack using the LoRa Basics Station mode.

#### How do you propose to document this?

...

#### Can you do this yourself and submit a Pull Request?

No. @nejraselimovic.

nejraselimovic, 2024-04-21:
@LDannijs any way you can get your hands on this gateway and write a doc?

LDannijs, 2024-04-22:
Sure!

PR #1340 (2024-06-09): Add Dragino LPS8N Basics Station docs

#### Summary

Add documentation for setting up the Dragino LPS8N with Basic Station 
Closes #1332 

#### Screenshots

[image]

#### Changes

- Add LBS docs for LPS8N

#### Notes for Reviewers

...

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is mar [... cut]
closed 2024-06-09 by nejraselimovic