TheThingsIndustries/lorawan-stack-docs #1346: Customized documentation for The Things Stack users based on Tenant ID and Cluster ID

KrishnaIyer, 2024-05-27:
#### Summary

[The Things Stack Documentation](https://www.thethingsindustries.com/docs/) is generic for everyone. The Things Stack Users are making mistakes with configurations. For example, to configure the migration tool with environment variables, we have given `eu1` as an example, but the users with the `nam1` cluster are configuring env variables with `eu1`. It would be great to have documentation with entering the user's tenant ID and cluster ID for customized links and configurations.

#### Why do we need this?

To have a documentation with customized links and configurations of The Things Stack Users based on Tenant ID and Cluster ID.

https://www.thethingsindustries.com/docs/

#### What do you want to see?

Upon accessing The Things Stack Documentation, a page with a dropdown asking for Deployment, Tenant ID and Cluster ID respectively, for customized documentation to different The Things Stack users with customized links and configurations. For example, CLI configuration, migration tool environment variables configuration, etc.

KrishnaIyer, 2024-05-27:
@pierrephz: This could be a nice design + implementation project for you.

pierrephz, 2024-06-25:
https://github.com/TheThingsIndustries/lorawan-stack-docs/assets/26456318/c21f39a5-d66f-46f5-996e-c57665c3fd3d

@KrishnaIyer find here a proposal for this new custom feature on the Docs

Add a toggle button in the header that opens a modal where you can fill your network info.

Note: I am also suggesting another structure for the header menu here. This is another topic but the current structure is messy and takes all the space of the header.

KrishnaIyer, 2024-06-26:
This is so cool. I love it. I would change the text to `Customize docs` but the rest looks great. Is this something that you can try to code yourself or would we need a dev for that?

kschiffer, 2024-06-26:
Some idea we had in the frontend meeting:

If the user is logged into SM/TTI, we could extract the tenant and deployment info and automatically apply it.