TheThingsIndustries/lorawan-stack-docs #1355: Floating "On this page" text on the right of the screen when you make the page horizontally larger

wienke, 2024-06-26:
[image]

KrishnaIyer, 2024-07-01:
This is the "Table of Contents" type section that's in most pages. This page does not have split contents so that's why it's empty. 
Maybe we should make this section conditional on the content. cc @nejraselimovic

nejraselimovic, 2024-07-07:
Yes, it would definitely look better if this wasn't shown when the table of contents is empty. I've looked into this but I don't know how to fix, can you @KrishnaIyer delegate it to someone who does?

I've also spotted some pages where the table of contents shouldn't be empty but it is because of wrong title sizes (like here https://www.thethingsindustries.com/docs/getting-started/lorawan-basics/), I will fix that part.