TheThingsIndustries/lorawan-stack-docs #1380: Misaligned/invisible distributions element

LDannijs, 2024-09-20:
#### Summary

The element that shows what distributions of The Things Stack are supported seems to be hidden and misaligned.  

https://github.com/user-attachments/assets/7476783f-17b4-4b4f-9719-0a7398b37cc7

#### Why do we need this ?

This needs to be fixed to properly show it.

#### What is already there? What do you see now?

The TTS distributions element is misaligned and appears invisible. The region of where it shows is also too large it seems.

#### What is missing? What do you want to see?

...

#### How do you propose to document this?

The dependencies got updated few days ago so i believe something went wrong with the update.

#### Can you do this yourself and submit a Pull Request?

Probably not, @johanstokking

KrishnaIyer, 2024-09-20:
We need to clean up this entire system of differentiating tiers/deployment models anyway.
@pierrephz: Let's work on some UX for this.

pierrephz, 2024-10-23:
### Here is a rework of the tags:

[image]

### Example of the tags in use:

[image]

ryaplots, 2024-10-31:
@KrishnaIyer There is also a "dedicated cloud" badge, what is this? Is this equivalent to enterprise or to cloud plus? Or do we not support this anymore?

KrishnaIyer, 2024-10-31:
Good question. We indeed don't use that anymore, you can remove references to those 👍

KrishnaIyer, 2024-11-11:
This is done via #1387 and deployed.

closed 2024-11-11 by KrishnaIyer