TheThingsIndustries/lorawan-stack-docs #1385: Minor changes to AWS ECS TTS deployment guide

SaiKumar-Bodapati, 2024-10-19:
#### Summary

The Things Stack user has requested that the documentation include the below Cloudformation templates in the AWS ECS Deployment guide.
  - `2-6-db-timescale-replica`
  - `2-7-queue-sqs` 
  - `5-5-tenant-resource-monitoring`
#### Why do we need this ?

- To help The Things Stack users understand the use case and instructions of the new AWS Cloudformation templates.

#### What is already there? What do you see now?

- Currently, we have descriptions and notes regarding all the available Cloudformation templates except the below:

  - `2-6-db-timescale-replica` (Instructions on this template are provided in the Template Change log : [Ref](https://www.thethingsindustries.com/docs/the-things-stack/host/aws/ecs/changelog/#3302) )
  - `2-7-queue-sqs` 
  - `5-5-tenant-resource-monitoring`

#### What is missing? What do you want to see?

Description related to the below AWS Templates on the AWS ECS deployment guide : [Link](https://www.thethingsindustries.com/docs/the-things-stack/host/aws/ecs/deployment/)

  - `2-6-db-timescale-replica`
  - `2-7-queue-sqs` 
  - `5-5-tenant-resource-monitoring`

#### How do you propose to document this?

- By adding the information related to the missing templates.

#### Can you do this yourself and submit a Pull Request?

No, @nejraselimovic

KrishnaIyer, 2024-10-21:
@happyRip. can you pick this one up?

KrishnaIyer, 2024-11-06:
I'll pick this up as part of a broader enterprise documentation update.

closed 2025-01-29 by KrishnaIyer