TheThingsIndustries/lorawan-stack-docs #1404: Missing embedded video in Adding Applications page

halimi, 2025-02-03:
#### Summary

On the `Integrations -> Adding Application` page should be an embedded video about how to add an application but the embedded video is not appearing, instead a HTML code is shown.

[image]

#### Why do we need this ?

To should be able to watch the video.

#### What is already there? What do you see now?

HTML code instead of the embedded video.

#### What is missing? What do you want to see?

I want to see the embedded video.

#### How do you propose to document this?

...

#### Can you do this yourself and submit a Pull Request?

...

KrishnaIyer, 2025-02-04:
Thanks for the issue. I'll take a look. Should be easy to fix.

sravankumar985, 2025-02-15:
@KrishnaIyer, in addition to [Adding Applications](https://www.thethingsindustries.com/docs/integrations/adding-applications/#adding-applications-using-the-console) what Halimi talked about, same is the issue with [Creating users](https://www.thethingsindustries.com/docs/concepts/advanced/user-management/user/#creating-users) and [Managing rights](https://www.thethingsindustries.com/docs/concepts/advanced/user-management/rights/#managing-rights-using-the-console)

##### Observations:

- On checking the docs repo, we did not see any difference between the section working fine([Docker](https://www.thethingsindustries.com/docs/enterprise/docker/)) and the one not working fine([Creating user](https://www.thethingsindustries.com/docs/concepts/advanced/user-management/user/#creating-users)).

- When we inspect the HTML elements, we see that the section that is not showing video properly has been encoded inside `<pre> </pre>`, and the div block which contains the iframe element is encoded within inverted commas as shown below and in the image.
```
<pre>
  <code>
    "<div> 
        <iframe ...> </iframe>
      </div>"
  </code>
</pre>
```
- But that is not the case with other sections where the video is rendered properly.

##### Supporting Internal Logs/Metrics

<details><summary>Image Comparing the docs code</summary>
<p>

[image]

</p>
</details> 

<details><summary>Images while inspecting the HTML section</summary>
<p>

[image]

[image]

</p>
</details>

KrishnaIyer, 2025-03-06:
Thanks for the inputs: this is fixed in https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/1412

PR #1412 (2025-03-06): Fix videos in tabs and add guide

#### Summary

Closes #1404

#### Screenshots

[image]

#### Changes

- Add a new shortcode to embed iFrames into tabs.
- Update contributing guide.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Run Locally: Verified that the docs build using `make server`, posted screenshots, verified external links. Test with `HUGO_PARAMS_SEARCH_ENABLED=true` if style changes will affect the search bar.
- [ ] New Features Marked: Documentation for new features is marked using the `new-in-version` shortcode, according to [... cut]